tomvandeneede/p2pp #164: Speed up 1st Layer FULLPURGEREDUCTION

dondec123, 2023-06-12:
The 1st tower layer using FULLPURGEREDUCTION is always printing at 20mms.  Any way to speed that up?    My coreXY machine does 1st layers at 55 no problem, the tower purge can be at least that fast.

Maybe there's an P2PP option for this I missed.  Thx in advance.

dondec123, 2023-06-12:
Here's the Prusa 3mf file, regular export, and Unprocessed.
https://1drv.ms/f/s!Arf1YeMUEh5Gm7o9J6SV_NWQ76JfVA?e=qwwYph 

Also, in theory (if I understand correctly) this 3 color print should produce the exact Color Object  with:
1. FULLPURGEREDUCTION on   -or-
2. PURGETOWERDELTA=999  -or-
3.  Both unused.

I'm just wondering if I need to verify this.   Thanks.

tomvandeneede, 2023-06-12:
Hmm in the file you sent me the first purge happens at 0.25mm

; --------------------------------------------------
; --- P2PP WIPE SEQUENCE START  FOR 208.28mm
; --- DELTA = 5.65
G1 X164.286 Y157.850 F8640
G1 Z0.250 F10800
G1 E0.50000

The actual height os the print at that time is 5.9mm

I will look into this further and see if I can find anything weird.

dondec123, 2023-06-13:
Hi Tom, have you figured out why its printing the 1st purge tower layer at 20mms?  My understanding is: with FULLPURGEREDUCTION enabled, its P2PP creating the purge tower, so in theory you have full control over its print speed.  

If that's the case, could you maybe create a new P2PP variable to allow user customization of the tower's 1st layer.    For example:  ;P2PP FULLPURGE1STLAYERSPEED=55 (for 55mm/s print speed).  Let me know.  Default 20mms is painful, lol.  Thanks.

tomvandeneede, 2023-06-14:
Change is not very straightforward but will see if I can fix something so set the minimum speed for the first layer

tomvandeneede, 2023-06-16:
9.0.3 is building and will be available shortly

just add the following line to your setup and you should be getting the normal purge speed throughout the whole tower

;P2PP FIRSTTOWERLAYERSPEEDUP

regards
Tom