tomvandeneede/p2pp #165: Error with single color prints

Killerskyman, 2023-06-26:
prusaslicer version: 2.6.0
p2pp version: 9.00.03
palette: palette 3 pro

When trying to print with palette with a single color the palette errors with error code '149'.

This is mainly in reference to #149 where i had the same issue and caused me quite a headache with a 12 hour print when it failed in the last 20 minutes. 

Having the same workflow and being able to configure the 'backup drives' for single color prints would be good and at minimum p2pp should at least give an error code beforehand that single color prints are not supported.

Killerskyman, 2023-06-26:
so it's actually like a firmware/software issue with the p3 itself then? wack.

tomvandeneede, 2023-06-26:
Iamaffraid so but maybe mosaic can solve in a next release

Killerskyman, 2023-06-26:
cool thanks for replying.

closed 2023-06-26 by Killerskyman