tomvandeneede/p2pp #166: Enhancement Requests...

dondec123, 2023-06-26:
Hi Tom, 
1.  When you knee deep in Splice Tuning woes, you even wonder if your P2PP Splice Table is taking effect (especially with lots of material combinations).  REQUEST: could you please PRINT the transition entries read in from Gcode and used for the current print, in the P2PP popup window: for example:  
W -to- RT : 1  0  3
RT -to- W : 1  1  3

2.  In the P2PP GCODE Splice Tuning section, could you please allow for the use of "white space" (spaces, tabs) instead of a single  _ underscore.  This would make the table far easier to read and to edit.

3. There's a bug which I slid past.  I created a material called "BLT" (blue transparent), and put that in the splice table.  But then  (as I recall from memory)  I had a material called "BL" (for black) listed after.  P2PP seemed to get confused and may have miss matched it to BLT.  I didn't save the warning messages for this, but this is a common text matching error... maybe just have a 2nd look at that code.

Thanks

      Don

tomvandeneede, 2023-06-27:
can you pls send me the code that triggers 3. an uprocesed gcode file would allow for testing as I have no issues on thos side bit I may be doing things differently or in a different order

Hi Tom,

dondec123, 2023-06-27:
Hi Tom,  Sorry I didn’t save that project.  All I remember were the details mentioned, and P2PP complaining that there was no splice method defined for that similarly named Color… when in fact it wasn’t even used in the print. If there’s nothing obvious… just ignore this.  Maybe user error.  Perhaps  I’ll rediscover it at some point and forward laster.  Sorry can’t be more helpful here.     Don From: Tom Van den EedeSent: Tuesday, June 27, 2023 12:35 AMTo: tomvandeneede/p2ppCc: dondec123; AuthorSubject: Re: [tomvandeneede/p2pp] Enhancement Requests... (Issue #166) can you pls send me the code that triggers 3. an uprocesed gcode file would allow for testing as I have no issues on thos side bit I may be doing things differently or in a different order

tomvandeneede, 2023-06-27:
Really a needle in a haystack to find what was happening…

9.0.4 should allow usage of spaces in the definition,
will look into printing the table of splices in a later release