tomvandeneede/p2pp #168: Enhancement Request: add to Splice Info

dondec123, 2023-06-29:
Hi Tom,
I'm in touch with the Mosaic Support, and it would be very helpful if P2PP would print the current Z Height and Layer number next to each entry in the Splice Information Table (if that's available in the software).

Thanks for your consideration.

    Don

tomvandeneede, 2023-06-29:
Hey don,

That’s going to be tricky… splices and code don’t really align physically.  
I can tell the z-height at which a splice happens in the print but this will be the total buffer length + tube length BEFORE it actually prints that filament

So not really sure what you are after

dondec123, 2023-06-29:
I see.  Haha... wow, nothing's easy here.   I get it.  Ok, just trying to make it less mystifying for debugging purposes.