tomvandeneede/p2pp #169: No pings generated in accessory mode

brex480, 2023-07-08:
Setup is Prusa slicer 2.6 , p2pp 9.00.5, P3pro feeding klipper. So far I have tried many different settings such as all types of purge and turning off other settings one by one. The only time it generates pings is when I disable accessory mode or when I get an error due to not setting the purge amounts in prusa slicer. I have also tried older versions of p2pp all the way back to 8.00.49
[image]

tomvandeneede, 2023-07-08:
can you please send me the jnprocessed gcode file so I can try tonfindntje cause

tomvandeneede, 2023-07-11:
> Thanks for taking a look at it. https://1drv.ms/f/s!AhAsS8sZJN-DgYBCAyMYm4dXAAGLLg?e=6d0eOF <x-msg://10/url>

looks like the unprocessed file is also a processed one.  can you please send the unprocessed file as it comes out of Prusa slicer before P2PP gets involved

thanks

brex480, 2023-07-11:
Add the ;P2PP SAVEUNPROCESSED correct? That's what I did the last time. Would temporarily just removing the p2pp.exe from the prusa slicer output section insure you get the correct file?

tomvandeneede, 2023-07-11:
both should yield the correct result.

brex480, 2023-07-12:
Sorry completely forgot about this by time I got home. Here is the file by removing p2pp from the output script to insure it's unprocessed. [https://1drv.ms/u/s!AhAsS8sZJN-DgYBRG91XH8Zdo-o0Ng?e=TUcHwJ](url)

tomvandeneede, 2023-07-13:
> Sorry completely forgot about this by time I got home. Here is the file by removing p2pp from the output script to insure it's unprocessed. https://1drv.ms/u/s!AhAsS8sZJN-DgYBRG91XH8Zdo-o0Ng?e=TUcHwJ <x-msg://12/url>

Hi,

currently the accessory mode only supports the standard tower processing, not the full purge processing.   Could definitely be added but requires quite some rework.

regards

brex480, 2023-07-13:
So to correct this do I just remove the purge tower settings from the start code? I believe I have tried that with the same results. Or is it a setting in prusa?

brex480, 2023-07-14:
Played around with it today and the issue was the unsupported purge tower. I believe when I was testing previously I never left out the purge tower and the auto add short splice at the same time. With those both left out of the start code it is working properly.

closed 2023-07-14 by brex480