tomvandeneede/p2pp #170: P2PP not communicating splice settings to Palette 3 Pro?

karmakina, 2023-07-16:
When running Palette 3 Pro with Canvas, or with its on on-board splice tuning, my splices are coming out fairly well.

But when running P2PP with PrusaSlicer, it's almost as if the Palette isn't being told how to properly splice. Everything seems to default to a low heat and compression value (which is no where in the gcode settings) even though the exact same settings work well on canvas and splice tuning.

When looking at the JSON file I noticed some interesting things, first it was using higher input numbers than were available (these were both on the same 4-color print generated twice in a row)

[image]
[image]

I hotfixed that by removing four of the inputs from the equation. This fixed the numbering system, but even though the algorithms were set up correctly in terms of numbers 1 through 4, they were mis-assigned in terms of which input > output combination matched which splice setting numbers. 

I then tried to adjust that by manually coding the correct input/output values and splice settings in the json file, recompressing it, and sending it to Palette. It accepted the file, but again it spliced every splice as if it had a zero heat value, despite the fact that the JSON file clearly shows this as the whole list of splices (this is a 3-color print):

[image]

Running that 3-color print through canvas was successful splicing. 

As far as I can tell, something is getting lost in translation. Prusa seems to be getting the code to P2PP just fine, but P2PP and the Palette 3 Pro are not communicating in terms of proper splice values. 

I'm not sure where to go from here. I'm normally pretty good at troubleshooting, but this one has me at a loss when my manual adjustments didn't stick and every splice broke in the tube.

I'm including the clean, unedited mcfx group file (as a zip file for the upload requirements). Maybe there's just something I'm missing that's small and easy to see that I can't figure out. 

[star-P2PP-original.zip](https://github.com/tomvandeneede/p2pp/files/12064700/star-P2PP-original.zip)

tomvandeneede, 2023-07-16:
Will have a look into this

tomvandeneede, 2023-07-16:
Can you please send me the unprocessed file so I can run here for debugging.

regards
Tom

karmakina, 2023-07-16:
Can you be more specific: As in the STL files? or project file? 

It's literally any file, I've tried 3 or 4 random ones at this point, and I even checked processed mcfx files from the past and found the same problem which maybe accounts for why I've been having so much difficulty with splicing working during tuning and not during printing.

tomvandeneede, 2023-07-16:
Code file generated by P2PP so I cnan run the script in debugging mode in the hope I can spot something

karmakina, 2023-07-16:
I can only provide the mcfx file in a zip file because of github 

[star-original-p2pp.zip](https://github.com/tomvandeneede/p2pp/files/12064816/star-original-p2pp.zip)

karmakina, 2023-07-16:
This is the file exactly as it appears straight out of P2PP (including .gcode)

[star-original-gcode.mcfx.zip](https://github.com/tomvandeneede/p2pp/files/12064825/star-original-gcode.mcfx.zip)

tomvandeneede, 2023-07-16:
But it is the code BEFORE p2pp ran right? Otherwise I cannot run it in deug mode as the file already has been processed.

karmakina, 2023-07-16:
I don't know what you're looking for, I'm sorry, I feel very dense right now. You said you wanted an unprocessed file, then the code generated "by" P2PP but now the code generated "before" P2PP. 

I need you to be more specific. The only file generated is by P2PP after the export g-code button is pressed, and I gave you that already.

karmakina, 2023-07-16:
Do you want the gcode file from inside the mcfx container? I can give you that, but it's odd because you could also get it by changing mcfx > zip and extracting, so I feel like that can't be it.

xile6, 2023-07-16:
Hi karmakina
Tom is looking for the gcode made when you add p2pp code 

;P2PP SAVEUNPROCESSED

This should save a unprocessed file

karmakina, 2023-07-16:
Thank you! This appears to be a .pp file saved to a temp directory, which I did not know about until now. 

Zipped and attached. 

[.28792.gcode_unprocessed.zip](https://github.com/tomvandeneede/p2pp/files/12065303/default.28792.gcode_unprocessed.zip)

karmakina, 2023-07-16:
Just to be clear, because I am frustrated we are going in circles for literally hours now, the .pp file appears to be the exact same text as the "print.gcode" file I included in the very first zip file in my very first post.

karmakina, 2023-07-17:
As a comparison, here are the files generated by the same file for use in Canvas. The JSON file in particular is correctly assigning algorithms (it labels the inputs as 12030000 and only identifies 1, 2 and 3 in its splice calculations), and the splices do come out well as I mentioned. (note that the 1:1 or 2:2 etc. splice settings are not defined as I wasn't using them so they default to 000, but it doesn't affect the print because as mentioned the assignments are correct, and it doesn't use any of these) 

(ignore the fact that the file says cube, it's literally the same star as the last one, canvas is annoying so I just use the same file over and over)

[cube1.zip](https://github.com/tomvandeneede/p2pp/files/12065536/cube1.zip)

the print code (unprocessed as it is an exploded mcfx file) is also there for Canvas.

tomvandeneede, 2023-07-17:
Processed your file just now and all appears to be adding up.   The filaments are mapped to materials (as they should as a same material could be used for multiple colours) and the materials map to the material splicing algorithms that are defined in the gone file:
;P2PP MATERIAL_Spark_Gold_1_3_7
;P2PP MATERIAL_Gold_Spark_1_3_7
;P2PP MATERIAL_White_Gold_1_4_7
;P2PP MATERIAL_Gold_White_1_4_7
;P2PP MATERIAL_Spark_White_1_2_7
;P2PP MATERIAL_White_Spark_1_3_7
;P2PP MATERIAL_Black_White_1_2_7
;P2PP MATERIAL_White_Black_2_3_7
;P2PP MATERIAL_Black_Spark_1_3_7
;P2PP MATERIAL_Spark_Black_1_3_7
;P2PP MATERIAL_Black_Gold_1_3_7
;P2PP MATERIAL_Gold_Black_1_3_7
;P2PP MATERIAL_DEFAULT_1_3_7

  "filaments": 
      "materialId": 3,      "filamentId": 1,      "type": "Gold",      "name": "YellowGreen",      "color": #DDC122"
      "materialId": 6,      "filamentId": 2,      "type": "Spark",      "name": "DodgerBlue",      "color": #8080FF"
      "materialId": 4,      "filamentId": 4,      "type": "White",      "name": "White",      "color": "#FFFFFF"

  "algorithms": 
      "ingoingId": 3,      "outgoingId": 6,      "heat": 1,      "compression": 3,      "cooling": 7
      "ingoingId": 3,      "outgoingId": 4,      "heat": 1,      "compression": 4,      "cooling": 7
      "ingoingId": 6,      "outgoingId": 3,      "heat": 1,      "compression": 3,      "cooling": 7
      "ingoingId": 6,      "outgoingId": 4,      "heat": 1,      "compression": 2,      "cooling": 7
      "ingoingId": 4,      "outgoingId": 3,      "heat": 1,      "compression": 4,      "cooling": 7

The numbers for the materials may be a little weird but they are taken from the following line in the gcode file:

; filament_type = Gold;Spark;Black;White;Cop;OWhite;OBlack;OGold

this line is split into components and then duplicates are removed.  This is a process that changes the order of the entries so you end up with the numbers 3,4,6 for the filament types
the order varies with each run so the material ID in two consecutive runs of the program can alter… but the processing is consistent.

Hope this answers your concern.

tomvandeneede, 2023-07-17:
BTW Canvas adds slicing onto the same type of filament but apparently this is not used in your print as it is not present in the code generated by P2PP
The values for self-splice are all 0,0,0 on canvas and are 1,3,7 from the default value in your p2pp file (should they occur)

karmakina, 2023-07-17:
Hi Tom,

You explained the unusual stuff in the JSON file, but I still have the same problem I did when I opened the ticket. Knowing why it's strange doesn't fix the problem.

P2PP + Prusa is still not splicing using the settings I've given it, whereas Canvas is, and Palette is (during splice tuning)

P2PP + Prusa is ignoring these values for some reason and using something of its own, I wouldn't know what settings it is using, all I know is that they are so brittle they can't stay together (so maybe 0/0/0 or 1/1/1)

karmakina, 2023-07-17:
Example splices: 

Top - P2PP (supposedly 1/3/7, but clearly not -- broke in the Palette's tubing)
Bottom - Splice Tuning (1/3/7) -- can bend without breaking

[image]

tomvandeneede, 2023-07-17:
what values donyou expect to be in the file?  I checked and the splice aloforithms reflect the valies you have specified.  I have no knowledge of the internalnprocessing so cannot speak to the devoce behavour but the file reflects what you specced in the settings … will take one more look when I am behind my machine tonight

karmakina, 2023-07-17:
All I can tell you is that it's not using the splice values I've set it up with, and you can see it for yourself int he picture. Since I don't code the application, I look to you for why that might be. 

If you can't fix it, I guess I will have to abandon the program and go back to canvas. Unless you have some other suggestions, solutions, settings, etc. all I can tell you is that P2PP + PrusaSlicer 2.6 is failing to tell Palette correct splice values and I can demonstrate it over and over again

tomvandeneede, 2023-07-17:
Did 9.1.0 which follows the numbering of the canvas tool. (still compiling but should be available in the next 30 mins or so)
It should not matter but as I don’t know the internals of the P3 something may have changed there in the last months making the alternative numbering causing an error

I double checked and all the splices are still defined in the file and correct according to your definitions.

karmakina, 2023-07-17:
I wish I could explain it or figure it out, but after something like 8 hours straight of troubleshooting on Saturday, I ran out of steam. I tried everything, including increasing splice values to ridiculous numbers like 5 or 6 compression, it didn't seem to budge. 

Maybe I should try factory resetting the palette or defining a new printer, new profile, hm.

tomvandeneede, 2023-07-17:
I have put 1000’s of hours into developing this tool and sorting out people’s issues.  I have a day time job and am in no way paid for any of this…

I see your frustration, but do that times 100 and this is how I feel about this right now.   Me too tlooked at your files for over 3 hours today trying to figure out what could possibly be wrong… and honestly I did not find anything that did not add up to a logical file structure… It could well be that mosaic in one of their latest updates changed something but unfortunately they will not tell me so my guess is as good as yours there… 

made me realise that it is better to stop p2pp altogether and pursue other ventures…

karmakina, 2023-07-17:
To be fair, no one else seems to be having my problem, so I'm still hoping it is somehow local to my setup. We'll see if I can figure it out!

karmakina, 2023-07-17:
I have a theory... we'll see if my theory plays out. (none of them have so far, lol)  -- I have a file on palette now starting to splice.

The theory is, that P2PP + Prusa combination is having trouble 'skipping an input', so it is kind of related to what we already said, but in a different way. So I've moved input 4 to input 3, since input 3 is not in use, and I'm going again.

karmakina, 2023-07-17:
I'm 4 splices in and it looks OK. It also explains everything: 1) why the problem hasn't ever happened before: because I never skipped an input until **this** print, and 2) why it wasn't affecting the first splice, which was a splice between only the first two inputs (didn't realize that until this afternoon).

So that's it. That's the problem. P2PP has trouble splicing if the inputs are not sequentially used 1 2 **0** 4 0 0 0 0 vs. if they are sequentially used 1 2 3 0 0 0 0 0, 3 filaments in both cases, only the second one splices correctly. 

You're doing good work Tom, I hope you don't give it up. I know it's thankless to do things for free, especially when people are annoyed. I spend my whole day troubleshooting other people's problems too, I do get prickly when I come home and have to do it for another 8 hours on something I do for fun, but I'll be honest, I still have fun in the end... as long as it gets solved eventually ;)

karmakina, 2023-07-18:
Completed one good print, and only one. After that, it went back to its original behaviour. Trying different variables to find out why, the only thing I changed were splice settings. It's definitely something about P2PP and the Palette 3 though...

karmakina, 2023-07-18:
I've got it going again after essentially taking the materialID and inputID and setting them to what I would expect them to be (1,2,3 for inputs 1,2,3 and material 1,2,3 for materials in inputs 1,2,3) -- it's working again. 

at least I'm having fun :)

tomvandeneede, 2023-07-18:
did you using 9.1.0?

karmakina, 2023-07-18:
Oh not yet, I will throw it on after work!

karmakina, 2023-07-19:
strangely enough, 9.0.1 did not solve the problem, but the code looks good, looks more like Canvas anyway.

So let me see if it's something else (such as, I don't have any self-splices defined in my code, but Canvas does, could it be that simple? Yes P2PP has 'default' but they aren't defined as self-splices in the algorithms)

karmakina, 2023-07-19:
Actually, wait, I found something more that's interesting. The json file with the algorithms is missing an entire algorithm! And the algorithm it is missing, is literally the second splice: 3 to 2. 

Why did P2PP miss an entire splice combination? Let me add it and I'll It is defined in the code. 

I'll add it manually, and run the project again. 

Here's the material definitions (I renamed them for ease of troubleshooting to 1 2 3 -- gold, blue, white)

```
;P2PP MATERIAL_PLA2_PLA1_1_3_5
;P2PP MATERIAL_PLA1_PLA2_1_3_5
;P2PP MATERIAL_PLA3_PLA1_1_4_5
;P2PP MATERIAL_PLA1_PLA3_1_5_5
;P2PP MATERIAL_PLA2_PLA3_1_2_5
;P2PP MATERIAL_PLA3_PLA2_2_3_5
;P2PP MATERIAL_DEFAULT_1_3_7
```

And this is the algorithms: 

```
"algorithms": [
    {
      "ingoingId": 1,
      "outgoingId": 2,
      "heat": 1,
      "compression": 3,
      "cooling": 5
    },
    {
      "ingoingId": 1,
      "outgoingId": 3,
      "heat": 1,
      "compression": 5,
      "cooling": 5
    },
    {
      "ingoingId": 2,
      "outgoingId": 1,
      "heat": 1,
      "compression": 3,
      "cooling": 5
    },
    {
      "ingoingId": 2,
      "outgoingId": 3,
      "heat": 1,
      "compression": 2,
      "cooling": 5
    },
    {
      "ingoingId": 3,
      "outgoingId": 1,
      "heat": 1,
      "compression":  #4,
      "cooling": 5
    }
  ]

```

Meanwhile the second color transition is literally 3 to 2 (white source/in and blue output) 

[image]

karmakina, 2023-07-19:
Confirmed fixed (once, at least): the problem was 1 missing splice definition (3 to 2). Once added manually, there are no issues. 

I will repeat the test but not tonight, possibly tomorrow.

tomvandeneede, 2023-07-19:
any chance sending me tjis fileweird though the it would still print if the splice definition wasmissing from the file

tomvandeneede, 2023-07-19:
> Ideally the problem could be solved by simply defining all available splices in material settings based on loaded inputs (including self-splicing based on default values or if values are defined) even if it won't be in use.
>
> Yes I will send the file after work, no problem

1/ I have identified in and out wrongly (they are sometimes confusing concepts  but I assume from output vid of the splices what goes in and what comes out….
this could explain why in your case you see 31 but not 13 in the file… anyway if 13 is missing then the file should not print and palette should give an error.
The reduction of number of splice definitions dates back from the P2 when the number of possible entries was physically limited.

2/ You don’t always have the necessary information to make the complete table. as such the program just selects the entries needed for completing the print… that is why I was surprised that I would have missed one and at the same time the P3 would still have done something as it would have absolutely no instructions on how to behave

3/ adding self-splices (material to same material) is the most standard one but if you have 3 different materials in the print and all splices will use different materials..no need to splice white to white or gold to gold or sparkle to sparkle as that will never happen… 

Concerning your star print where you say a splice type seems missing would you mind doing a test with the P2PP output?

take the splce definition that you thing is the opposite way around :  I believe it is the following one
    {
      "ingoingId": 2,
      "outgoingId": 3,
      "heat": 1,
      "compression": 2,
      "cooling": 5
    },

change it as follows: (change, not copy and add, so the 2,3 splice pattern is no longer in the file

    {
      "ingoingId": 3,
      "outgoingId": 2,
      "heat": 1,
      "compression": 2,
      "cooling": 5
    },

then try to print the file and see if it complains for the missing splicing info.   There is no use for me adding the 3,2 in the file as the issue would clearly be in the interpretation of ingoing and outgoing filament… At least if p3 gives an error it would be clear that the orderer seems correct.   If it does not give an error is does nog prove anything really.  Most people print with single type so the problem will not be widespread and could potentially but a quick fix if it is just swapping the I/O numbers

Tom

tomvandeneede, 2023-07-19:
would you mind testing 9.1.1

karmakina, 2023-07-19:
will do!

karmakina, 2023-07-19:
It looks good! I will run another test, something different, just to be sure as soon as I'm not dying of the heat for just a second.

karmakina, 2023-07-20:
The splice settings continue to be wrong I'm afraid. 

This is my newest code algorithms:

```
 "algorithms": [
    {
      "ingoingId": 2,
      "outgoingId": 1,
      "heat": 1,
      "compression": 5,
      "cooling": 5
    },
    {
      "ingoingId": 3,
      "outgoingId": 1,
      "heat": 1,
      "compression": 5,
      "cooling": 5
    },
    {
      "ingoingId": 1,
      "outgoingId": 2,
      "heat": 1,
      "compression": 3,
      "cooling": 5
    },
    {
      "ingoingId": 3,
      "outgoingId": 2,
      "heat": 1,
      "compression": 2,
      "cooling": 5
    },
    {
      "ingoingId": 1,
      "outgoingId": 3,
      "heat": 1,
      "compression": 4,
      "cooling": 5
    }
  ]
```

vs. the PrusaSlicer definitions on materials 

```
;P2PP MATERIAL_PLA2_PLA1_1_3_5
;P2PP MATERIAL_PLA1_PLA2_1_5_5
;P2PP MATERIAL_PLA3_PLA1_1_4_5
;P2PP MATERIAL_PLA1_PLA3_1_5_5
;P2PP MATERIAL_PLA2_PLA3_1_2_5
;P2PP MATERIAL_PLA3_PLA2_2_3_5
;P2PP MATERIAL_DEFAULT_1_3_7
```

They still don't match. PLA2 (on input 2) > PLA1 (input 1, becomes output here) should be 1 3 5 but your code indicates it is 1 5 5. In reality this is meant to be PLA1 > PLA2 as 1 5 5. Meanwhile PLA3 (on 3) > PLA2 (on 2) is defined as 1 2 5 but should be 2 3 5

So everything is definitely more backwards than before sadly. It seems like the last file was not backwards, but was simply missing an entire splice algorithm.

karmakina, 2023-07-20:
Sure :) Do you want for 9.01.0 or 9.01.1 ?

karmakina, 2023-07-20:
Sure! Let me load the star again, it is easiest to diagnose due to how predictable all the colored parts are. 

9.01.00 -- this is the one that is most correct in most splices, but is missing the 3 > 2 transition which is literally the second splice

[star-90100.zip](https://github.com/tomvandeneede/p2pp/files/12111361/star-90100.zip)

9.01.01 -- has the 3 > 2 transition, but everything else is backwards. 

[star-90101.zip](https://github.com/tomvandeneede/p2pp/files/12111364/star-90101.zip)

tomvandeneede, 2023-07-20:
will likely be somewhere tomorrow when I look into this, hase visitors tonight,

karmakina, 2023-07-20:
No rush at all, I have it working on manual adjustment which isn't difficult for me to do :)

dondec123, 2023-07-20:
Following this discussion.  I was having trouble with splices too, spending weeks (I'm retired, so have the time), trying to figure what was going wrong.  I asked Tom for more visible debugging information in the P2PP compilation window, specifically what splice values were used at each splice (e.g, W to B 1 3 4, etc).  I was hoping this would be added to help diagnose problems like those discussed here.   At this point I'm not sure ver 9.01.01 fixes or breaks P2PP.   I hate to use Canvas.  Is the recommendation to use only successive input channels (1,2,3) as opposed to (1,2,5)?  Or is the problem more complicated than that?   I have a 4 color print running tonite on P3P device.  Thanks.

karmakina, 2023-07-20:
> 9.01.01

Don't use  9.01.01, it reverses everything. I'm using 9.01.00 and checking the json myself (adding in any splice value I fear might be missing) and repackaging, deploying. Works fine.

I haven't tested unfortunately skipping inputs on the newer version, if you intend to queue up a longer print, if it were me, I'd load them successively. But I'd also double check the json file that all the splice transitions are listed in the algorithms. Currently, as I can't always know for 100% sure a missing splice will be required (sometimes you might have 3 > 1 and never ever have a 1 > 3, but can you know for sure? I think it's hard to tell), I am just adding all of my splice settings in as algorithms.

karmakina, 2023-07-20:
I ran a quick slice with a skipped input and this is what it looks like on 9.01.00 

[image]

This is also what Canvas does, so that matches what Mosaic is doing. Therefor skipping inputs _in theory_ should not be an issue, but note I did not test that theory. And it doesn't mean that you will have all your splices defined in the algorithms. To be 100% safe, add all possible splice transition settings to algorithms.

dondec123, 2023-07-20:
Ok, I'm a retired programmer, but not sure how to unpack and manually check the code.  Do I enable the P2PP debugging option (P2PP DEBUGTCOMMAND or SAVEUNPROCESSED)... is that how you get to the JSON code.  After checking it (possibly modifying it), how to repack and transmit to Palette?   Thanks.

karmakina, 2023-07-20:
Take your MCFX file, change the extension from .mcfx to .zip, and unpack it. Four files will be there, and the one with the algorithms defined is palette.json

[image]

When you're ready, re-zip the files (not the whole folder, just the 4 files) and name it back to mcfx

dondec123, 2023-07-21:
Ok, very clear.  I'm so happy you had the skill to diagnose this issue.  I wasn't getting anywhere, I had no ideal about the JSON code and zip packing.  This is a huge step in the right direction.   Thank you.

dondec123, 2023-07-21:
I just verified that 9.01.00 and 9.01.01 are both missing splice settings in my own 4 color print.  And as karmakina pointed out, the ingoing and outgoing ID values are reversed in 9.01.01.

tomvandeneede, 2023-07-21:
does eirher of the versions male p3 raise an error?  the settings are indeed reversed in botj versions and one of the definitely misses a setting…. so one of them should be containing the correct information the other shouldnmiss one.   Prusaslicer always statts the next lauer in the color is ended with and then goes back so there is most of the time one splice algontjat isnot needed or used and is  ot represented in the target file.

karmakina, 2023-07-21:
9.01.00 is the best right now except missing splice information, it is not reversed but is missing splice algorithm data. 

You should take 9.01.01 off the website if you haven't already, that one is unusable. 

There are no Palette errors. 

Even though you're saying one splice is not needed, I've proven that to be false, and a simple solution is just to define all available splice information in the algorithms.

tomvandeneede, 2023-07-21:
9.1.1 should be with the one splice missing reversed so it is not recersed just missing  912 should add the missing splice that is not needed

karmakina, 2023-07-21:
9.1.1 is fully reversed

9.1.0 is normal except for a missing splice

tomvandeneede, 2023-07-21:
Checking with mosaic people

Seems that eels-splicing is needed in case you run out of filament and need to splice onto yourself
They say they also splices that are not used need to be in (this is a new requirement) so I will put them in as well
Should be releasing 9.1.3 with this later tonight

tomvandeneede, 2023-07-21:
9.1.3 is compiling now should have all possible permutations for the filaments used. 
NOTE: you should also define self-splices in case you have a runout so P3 knows how to attach filament to the same type of material

karmakina, 2023-07-21:
got it. I'll put my next print on with 9.1.3 when it's ready!

tomvandeneede, 2023-07-21:
should be there now

karmakina, 2023-07-21:
I only see 9.1.2

tomvandeneede, 2023-07-21:
Will check in a minute

tomvandeneede, 2023-07-21:
Was still uploading, is there now

[19 more replies]