tomvandeneede/p2pp #171: incorrect starting layer height in P2PP-generated gcode file

bavery79, 2023-07-23:
FYI, last few versions of P2PP do something weird to the z offset and prints are starting about 7mm above the bed for first layer.  9.00.04 is the latest version I can get to actually successfully generate gcode without this issue. Currently using PrusaSlicer 2.6.0 rc2, if that is relevant. Thanks.

tomvandeneede, 2023-07-24:
can you please send me the unprocessed gcode file for thisprint so I can have a look

bavery79, 2023-07-25:
Please try this.  Not sure if the file will work since appears too large to
attach directly.
 BRYAN - ArtFlex Baby Peacock__REPRINT___PP2P_9....
<https://drive.google.com/file/d/1hyOwsZ_99uH0AeOnioETxSDBbLWyjzTU/view?usp=drive_web>

bavery79, 2023-07-25:
[missing G1 z-moves .zip](https://github.com/tomvandeneede/p2pp/files/12158789/missing.G1.z-moves.zip)

Here you go. Included gcode generated by both 9.00.04 and 9.01.04 versions.  For some reason it appears with the newer version the G1 lines containing z-moves are completely missing so it looks like the print is starting from whatever z-height my klipper START_PRINT macro ended up after homing.  Hope this helps! Thank you.

tomvandeneede, 2023-08-06:
All I see is a missing G1 Z-moves.zip but it contains processed files, not the unprocessed file I can run through P2PP myself

bavery79, 2023-08-08:
Oh I see. Will try again then. :-)

tomvandeneede, 2023-08-10:
Let me know when it is there… really curious about this

hviet17, 2023-08-11:
hello, I'm having the same issue as well. Ver 9.00.04 works fine for me. Additionally somehow the script doesn't use relative E distance ("Move exceeds maximum extrusion" error). Here's my file generated by version 9.01.04
[.22712.gcode_unprocessed.pp.zip](https://github.com/tomvandeneede/p2pp/files/12321940/default.22712.gcode_unprocessed.pp.zip)

My start gcode:
```
;P2PP PALETTE3_PRO
;P2PP P3_PROCESSPREHEAT
;P2PP PRINTERPROFILE=adb063623f59d256fa0431d9171be49f
;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_0_0_0
;P2PP LINEARPINGLENGTH=350
; Adapt to the length of your bowden tube
;P2PP EXTRAENDFILAMENT=150
;P2PP ACCESSORYMODE_MAFX
;P2PP KLIPPER_TOOLCHANGE

SET_GCODE_VARIABLE MACRO=START_PRINT_MULTI VARIABLE=bed_temp VALUE=[first_layer_bed_temperature]
SET_GCODE_VARIABLE MACRO=START_PRINT_MULTI VARIABLE=extruder_temp VALUE=[first_layer_temperature]
START_PRINT_MULTI
```

tomvandeneede, 2023-08-11:
9.1.5 should resolve this issue

bavery79, 2023-08-12:
Yes, looks like the update solved the issue for me!