tomvandeneede/p2pp #172: Enhancement Requests - Accessory Mode & P2PP-controlled wipe towers

bavery79, 2023-07-31:
Moving from P2 to P3P (with a Klipper-based printer) recently, I find myself missing a lot of the extra functionality allowed by the FULLPURGEREDUCTION mode that I had been using regularly with my Palette 2S.  Ran across a closed issue regarding Accessory Mode not able to support any of the P2PP-generated wipe tower modes and would like to add my 2-cents.  

On a related note, I would love to see an ability to take advantage of those extra functions I've enjoyed (specifically WIPEFEEDRATE & AUTOADDPURGE) in something like the tower delta mode (ideally with accessory mode support for klipper but I understand this circles back to request #1 above), so a user could for example set a delta of 0 and then be able to get a tower built right next to the printed model while still being able to use AUTOADDPURGE and WIPEFEEDRATE features.  Or maybe it could be as simple as - if you try to enable one of those features with the default Prusa-generated wipe tower P2PP could just auto-switch to generate its own tower.  Basically, anything to make AUTOADDPURGE useable with a more "default" setup of P2PP I think would add greatly to useability and give one less thing to have to worry about when setting up prints.

tomvandeneede, 2023-08-08:
will do some further investigation to see if these can be enabled without major rewrites of the code

bavery79, 2023-08-08:
I see I was apparently mistaken about autoaddpurge and it actually dies already work in the default mode with prusa generated wipe tower. Could have sworn this wasn't the case a few months ago, but might have been user error on my part. :-)

bavery79, 2023-08-12:
> I see I was apparently mistaken about autoaddpurge and it actually dies already work in the default mode with prusa generated wipe tower. Could have sworn this wasn't the case a few months ago, but might have been user error on my part. :-)

Nevermind, it seems I spoke too soon!  New PP2P version now shows a confirmation that autoaddpurge has been disabled due to incompatibility.  Perhaps the older PP2P version just wasn't showing the notification?

tomvandeneede, 2023-08-14:
Autoaddpurge can only be done with self-genrrated purge towers.  I need control over where to purge the filament 
Will think about a way to resolve this but nothing comes to mind right now