tomvandeneede/p2pp #173: p2pp generate gcode different from preview for wipe tower

hviet17, 2023-09-06:
I had this issue recently. When generating gcode with p2pp, in 1st layer, there's a brim around wipe tower that overlap the skirts & other parts. I think it also messes with the color change. In PrusaSlicer this doesn't show up. I tried print gcode without p2pp and it printed correctly like preview. 

[image]

With p2pp:
[image]
Without:
[image]

Gcode:
[.15604.gcode.pp.zip](https://github.com/tomvandeneede/p2pp/files/12534191/default.15604.gcode.pp.zip)

tomvandeneede, 2023-09-06:
can you please send me unprocessed code for this print then I can take a look.

hviet17, 2023-09-06:
@tomvandeneede I closed the project & can't recreate this issue anymore. Will come back here when I come across it again. Thank you.