tomvandeneede/p2pp #174: Possibly wrong splice setting when printing start from input 8

hviet17, 2023-09-10:
I had a print that use input 8,7,6 & 2, other inputs are skipped. Somehow the 1st splice on the print doesn't look like what I had when spice tuning. My relevant settings:
```
;P2PP MATERIAL_DEFAULT_-1_0_1
;P2PP MATERIAL_PLA_PLA_-1_0_1

;P2PP MATERIAL_PLA6_PLA2_-2_3_3
;P2PP MATERIAL_PLA2_PLA6_-2_3_3
```

I think the generated gcode use the fallback value instead of what I defined. I don't know if I'm doing something wrong. Any help would be much appreciated.
[image]
[image]
[.12636.gcode_unprocessed.pp.zip]
Gcode:
(https://github.com/tomvandeneede/p2pp/files/12568585/default.12636.gcode_unprocessed.pp.zip)

tomvandeneede, 2023-09-10:
Which version of P2PP are you using?

hviet17, 2023-09-10:
@tomvandeneede I'm using the latest 9.01.05

tomvandeneede, 2023-09-11:
can you please send me the processed and unprpcessed files so I can take a look

(https://github.com/tomvandeneede/p2pp/files/12568585/default.12636.gcode_unprocessed.pp.zip)

hviet17, 2023-09-11:
@tomvandeneede here're the files
[gcodes.zip](https://github.com/tomvandeneede/p2pp/files/12574949/gcodes.zip)

hviet17, 2023-09-12:
@tomvandeneede I found the culprit. Turn out FILAMENTOVERRIDE is outdated & I have to use custom filament type in Filament Setting. Thank you for your time.

closed 2023-09-12 by hviet17