tomvandeneede/p2pp #177: P2PP Support for PrusaSlicer on Linux

N3rdB01-1, 2023-12-09:
I notice that there is not a support direct for linux. Is there one in the works? I tried to build from source but seems to not like something about my setup.

tomvandeneede, 2023-12-09:
Should work I guess…. I use one of the standard packaging techniques, it is a matter of creating the config and compile.
I know other people are  running on linux from the source tree, but you need to make sure you get all the prerequisites right.

What error are you getting?

N3rdB01-1, 2023-12-11:
There were several missing dependencies. It still doesn't seem like something is correct. I don't know of I am supposed to see an option in prusaslicer or not. I can't find the multi material usage button ( that is related to prusaslicer I am aware) nor do I see an option to post process the file with p2pp.

matiasdj87, 2024-01-24:
Use with WINE in terminal.. wine p2pp.exe GCODEfile.gcode