tomvandeneede/p2pp #178: Big blobs in big prints if FULLPURGEREDUCTION is enabled

xTremePower, 2023-12-14:
Hello,

Thanks for all your work (i just tipped you10€ :p ) 

It's the first time this problem happens to me :
I tried to print one of my large color design, 
I tried 2 times because the first time I thought it was the printer's fault, but even after slicing gain with the purge tower and model rotated 90° it give same problem (just not at same place)

After many tries, I think about FULLPURGEREDUCTION, I disabled it, slice again the Same model, same place same setting, I only removed the FULLPURGEREDUCTION from start gcode, and it worked like a charm...

As you can see on the pictures, It put big blobs from I dont know where, or scratch the side of the printed object, just removing FULL remove totally this problem ! I don't know how and why, but now it works

But now the problem without FULL is : P2PP give an error, saying Automatic Splice length increase works with Full purge reduction and side wipe only, so if i have some short splices, it will not work :(

I only have the mcfx files of both slice, what can i give you to help you ?

Thanks again

All pictures are with FULL enabled, if disabled this big blobw don't exist at all
[image]
[image]
[image]
[image]
[image]
[image]

xTremePower, 2023-12-15:
PS : I forgot one other important thing : 

When I print without FULLPURGEREDUCTION, the print always start on the purge block, it's perfect because it prime and clean the nozzle,
But when I print THE SAME model same settings etc, only FULLPURGEREDUCTION enabled, the print start directly on the model, so many times the first centimeters are missing

tomvandeneede, 2023-12-15:
Will try to take a look in the morning

tomvandeneede, 2023-12-16:
OK I will need some extra info to look into this.

Can you please send me the unprocessed file for the print with the errors.   lso lat me know which version of PS you are using… maybe something has changed that I am not aware of

regards

xTremePower, 2023-12-16:
Hello buddy,

How can I obtain the unprocessed file ? You speak about the 3mf PrusaSlicer project file ?

I use the last version 9.01.05

Thanks

tomvandeneede, 2023-12-16:
not the 3mf,
just replace and save with all the same settings but temp remove the p2pp processing command (or put an extra p (like pp2pp) in the command so it does not execute.  The gcode file you get then is what I would need

xTremePower, 2023-12-16:
ah okay :p 
[notGood_13h32m_375.624gr_H0.25_N0.6.gz](https://github.com/tomvandeneede/p2pp/files/13692694/notGood_13h32m_375.624gr_H0.25_N0.6.gz)
Here are the file, it's a gcode renamed gz to let github to uplad it

tomvandeneede, 2023-12-20:
Hi Zoltan

got the file now and trying to find where it goes wrong…  not yet found where it is stalling

Hotwire, 2024-01-25:
Just wanted to add, I am having this issue too, but on the purge block. there are some big globs happening.

tomvandeneede, 2024-01-26:
Are you using the accessory mode or the standard P3 ping mode?

xTremePower, 2024-04-28:
I use the normal connected mode, the palette drives the printer directly