tomvandeneede/p2pp #179: P2pp with palette 3 pro and K1 Max

Fredleonard1, 2023-12-17:
Hello P2PP is configure correctly and send good code to the palette 3 pro. When I want to start print palette 3 Pro does not recognize the printer so ask to connect it. The K1 Max can not connect at this time .

- Try to configure the printer in Canvas and it work in accessory mode. Is there a command in P2PP to tell that the palette 3 pro will be in accessory mode ?
- Try to enter the ID code of the printer since I have already configure this printer in Canvas . Was unable to find in Canvas or in the palette pro the ID number with 32 digit as it need to be program in P2PP. 
- I use the Id of the printer that i have find in Creality Cloud but it is a 14 digit ID and P2PP tell me that we need a 32 digits ID. 
- So I can have the palette 3 pro work with K1 max but only with canvas . Unable to make it work with Prucaslicer.

tomvandeneede, 2023-12-18:
there is a section in the p3 manual describing how to set this up

Fredleonard1, 2023-12-18:
Ok thanks the when I load the PDF on the site it as a bug and not showing the end of the document where this info was available . I have download the doc document and now I see the info. Thanks for the quick response.

closed 2023-12-18 by Fredleonard1