tomvandeneede/p2pp #180: Warning: Variable layers may cause issues with FULLPURGE / TOWER DELTA

Fatihhamircu, 2023-12-18:
Dear all,

I also printed different products on the palette 3 pro and it works. And now, I downloaded and installed p2pp to use the features of Prusa slicer.

I have two different colored objects in Scene. These are molar tooth and nerve inside of it. After the slice, I take the warning below.

```
Variable layers may cause issues with FULLPURGE / TOWER DELTA
This warning could be caused by support that will print on variable layer offsets
```

What had I better do?

Thanks in advance 

Best Regards

NOTE:
My Setting:

```
G21 ; metric values
G90 ; absolute positioning
M82 ; set extruder to absolute mode
;P2PP PALETTE3_PRO
;P2PP PALETTE3_PRO
;P2PP PRINTERPROFILE=12345678901234567890123456789012

;P2PP SPLICEOFFSET=40

;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90

;P2PP EXTRAENDFILAMENT=150
;P2PP LINEARPINGLENGTH=350

;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_2_1_2

;P2PP ABSOLUTEEXTRUDER

;P2PP FULLPURGEREDUCTION 
;P2PP WIPEFEEDRATE=3000 
;P2PP AUTOADDPURGE 
```

What I got from compiler:

```
P2PP Version: 9.01.05
Python Version: 3.9.1
Platform: win32

Reading File C:\Users\USER\AppData\Local\Temp\.12740.gcode.pp
Analyzing Prusa Slicer Configuration
All purge lenghths 70/70 OR 140. Purge lengths may not have been set correctly.
SPLICE OFFSET: 40.00mm
Extra filament at end of print 150.00mm
Linear Ping interval of 350.00mm
Convert to absolute extrusion parameters
Full purge reduction configured
GCode Analysis ... Pass 1
Tower detected from (7.537999999999999, 43.852) to (57.294000000000004, 72.888)
Variable layers may cause issues with FULLPURGE / TOWER DELTA
This warning could be caused by support that will print on variable layer offsets
FULLPURGEREDUCTION activated
Automatic Splice length increase activated
Gcode Analysis ... Pass 2
Processing time 4.45s
Generating MCFX file: C:\Users\USER\AppData\Local\Temp\.12740.gcode.pp
Generating Palette 3 output files
```

tomvandeneede, 2023-12-18:
When using variable layers, turn off the FULLPURGEREDUCTION.  They are incompatible since the FULLPURGEREDUCTION layers are calculated on a fixed layer height… they will likely not be in sync with the print and potentially cause collisions.

Fatihhamircu, 2023-12-18:
> When using variable layers, turn off the FULLPURGEREDUCTION. They are incompatible since the FULLPURGEREDUCTION layers are calculated on a fixed layer height… they will likely not be in sync with the print and potentially cause collisions.
> […](#)

how to close variable layers?