tomvandeneede/p2pp #182: Printing two objects

Fatihhamircu, 2023-12-25:
I have molar tooth and nerves of it. I have extracted them from Cone Beam Tomography and other softwares.
[image]
[image]

I have **scaled up** these objects.
I have intented to print out tooth as **transparent** and nerve as **red** with pla.
I have used lightening infill with %15 to hold the nerve in the air(seems to be in the air).

**My Gcode is;**
`G21 ; metric values
G90 ; absolute positioning
M82 ; set extruder to absolute mode
;P2PP PALETTE3_PRO
;P2PP PALETTE3_PRO
;P2PP PRINTERPROFILE=12345678901234567890123456789012

;P2PP SPLICEOFFSET=40

;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90

;P2PP EXTRAENDFILAMENT=150
;P2PP LINEARPINGLENGTH=350

;P2PP MATERIAL_DEFAULT_1_2_1
;P2PP MATERIAL_PLA_PLA_2_1_2

;P2PP ABSOLUTEEXTRUDER

;P2PP SIDEWIPELOC=X254 
;P2PP SIDEWIPEMINY=45 
;P2PP SIDEWIPEMAXY=195 
;P2PP SIDEWIPECORRECTION=1.0 

;set !P2PP SIDEWIPELOC=X254 
;set !P2PP SIDEWIPEMINY=45 
;set SIDEWIPEMAXY to the same value as SIDEWIPEMINY 
;set !P2PP SIDEWIPEMAXY=45 
;set WIPEFEEDRATE is required to override default value!! 
;set !P2PP WIPEFEEDRATE=200  

;P2PP PURGETOWERDELTA=0 

;P2PP FULLPURGEREDUCTION 
;P2PP WIPEFEEDRATE=3000
;P2PP AUTOADDPURGE 
 

M140 S[first_layer_bed_temperature] ; set bed temp
G28 X0 Y0 ; home X and Y
G28 Z0 ; home Z
G1 Z30 F{machine_max_feedrate_z[0]*60} ; move Z a bit down to not blow on the bed edge while heating
G1 X10 F3900 ; let some space on x to prevent the filament cooling exhaust from beeing blocked by the servo motor
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M106 S80 ; turn on fan to prevent air nozzle melt while heating up
M109 S[first_layer_temperature] ; wait for extruder temp
M107 ; start with the fan off
G28 X0 ; goto X home again
G92 E0 ; zero the extruded length
G1 Z0.2 F360 ; move plattform upwards
; extrude material next to the plattform (comment or remove following lines to disable)
G1 F180 E20 ; extrude some material next to the plattform
G92 E0 ; zero the extruded length
G1 E-[retract_length] F{retract_speed[0]*60} ; do a filament retract
G92 E0 ; zero the extruded length again
G1 X5 F3900 ; move sideways to get rid of that string
G1 E[retract_length] F{retract_speed[0]*60} ; do a filament deretract with retract parameters
G92 E0 ; zero the extruded length again
; draw intro line (comment or remove following lines to disable)
G1 X30 E5 F700 ; draw intro line
G92 E0 ; zero the extruded length
G1 E-[retract_length] F{retract_speed[0]*60} ; do a filament retract
G1 X40 Z2.0 ; move away from the introline
G92 E0 ; zero the extruded length again
G1 E[retract_length] F{retract_speed[0]*60} ; do a filament deretract with retract parameters
; end of intro line code
M117 Printing...
G5`

When I have run the prusa slicer with p2pp, I got the below **preview** of 3d print:
[image]

 **What I got from printer is;**
[image]
[image]

**My question is:**
Why the molar tooth and infill is colored even if I selected the extruder 1 which is transparent PLA for molar tooth?

Note;
I have palette pro 3 and anycubic 4max pro 2 as hardware and p2pp and prusaslicer as software.

tomvandeneede, 2023-12-25:
This is weird

Could you send me the unprocessed gcode
Which version of Prusa Slicer are you using?

regards
Tom

Fatihhamircu, 2023-12-25:
UPDATE:
My bundle is:
`# generated by PrusaSlicer 2.7.1+win64 on 2023-12-25 at 17:27:15 UTC

[print:0.20mm Quality @4Max Pro 2.0 -  printSetting - mosaic palette]
arc_fitting = disabled
avoid_crossing_curled_overhangs = 0
avoid_crossing_perimeters = 0
avoid_crossing_perimeters_max_detour = 0
bottom_fill_pattern = concentric
bottom_solid_layers = 4
bottom_solid_min_thickness = 0
bridge_acceleration = 300
bridge_angle = 0
bridge_flow_ratio = 0.65
bridge_speed = 30
brim_separation = 0
brim_type = outer_only
brim_width = 5
compatible_printers = 
compatible_printers_condition = printer_model=="4MAXPRO20" and nozzle_diameter[0]==0.4
complete_objects = 0
default_acceleration = 900
dont_support_bridges = 1
draft_shield = disabled
elefant_foot_compensation = 0.2
enable_dynamic_overhang_speeds = 0
external_perimeter_acceleration = 0
external_perimeter_extrusion_width = 0
external_perimeter_speed = 25
external_perimeters_first = 0
extra_perimeters = 0
extra_perimeters_on_overhangs = 0
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0.45
fill_angle = 45
fill_density = 15%
fill_pattern = lightning
first_layer_acceleration = 300
first_layer_acceleration_over_raft = 0
first_layer_extrusion_width = 0.65
first_layer_height = 0.2
first_layer_speed = 20
first_layer_speed_over_raft = 30
fuzzy_skin = none
fuzzy_skin_point_dist = 0.8
fuzzy_skin_thickness = 0.3
gap_fill_enabled = 1
gap_fill_speed = 40
gcode_comments = 0
gcode_label_objects = firmware
gcode_resolution = 0.0125
gcode_substitutions = 
infill_acceleration = 600
infill_anchor = 400%
infill_anchor_max = 50
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_overlap = 23%
infill_speed = 45
inherits = 0.20mm Quality @4Max Pro 2.0
interface_shells = 0
ironing = 0
ironing_flowrate = 15%
ironing_spacing = 0.1
ironing_speed = 15
ironing_type = top
[... 2107 lines cut ...]
travel_max_lift = 0,0,0,0,0,0,0,0
travel_ramping_lift = 0,0,0,0,0,0,0,0
travel_slope = 0,0,0,0,0,0,0,0
use_firmware_retraction = 0
use_relative_e_distances = 1
use_volumetric_e = 0
variable_layer_height = 0
wipe = 1,0,0,0,0,0,0,0
z_offset = 0

[presets]
print = 0.20mm Quality @4Max Pro 2.0 -  printSetting - mosaic palette
sla_print = 
sla_material = 
printer = Anycubic 4Max Pro 2.0 - printerSetting - palette 3 pro
filament = Generic PLA @4Max Pro 2.0 - PLASetting - mosaic palette
filament_1 = Generic PLA @4Max Pro 2.0
filament_2 = Generic PLA @4Max Pro 2.0
filament_3 = Generic PLA @4Max Pro 2.0
filament_4 = Generic PLA @4Max Pro 2.0
filament_5 = Generic PLA @4Max Pro 2.0
filament_6 = Generic PLA @4Max Pro 2.0
filament_7 = Generic PLA @4Max Pro 2.0
physical_printer = 
`

tomvandeneede, 2023-12-25:
Would you mind attaching the full unprocessed code?

> On 25 Dec 2023, at 18:34, Fatihhamircu ***@***.***> wrote:
> 
> 
> UPDATE:
> My bundle is:
> `# generated by PrusaSlicer 2.7.1+win64 on 2023-12-25 at 17:27:15 UTC
> 
> [print:0.20mm Quality @4max <https://github.com/4max> Pro 2.0 - printSetting - mosaic palette]
> arc_fitting = disabled
> avoid_crossing_curled_overhangs = 0
> avoid_crossing_perimeters = 0
> avoid_crossing_perimeters_max_detour = 0
> bottom_fill_pattern = concentric
> bottom_solid_layers = 4
> bottom_solid_min_thickness = 0
> bridge_acceleration = 300
> bridge_angle = 0
> bridge_flow_ratio = 0.65
> bridge_speed = 30
> brim_separation = 0
> brim_type = outer_only
> brim_width = 5
> compatible_printers =
> compatible_printers_condition = printer_model=="4MAXPRO20" and nozzle_diameter[0]==0.4
> complete_objects = 0
> default_acceleration = 900
> dont_support_bridges = 1
> draft_shield = disabled
> elefant_foot_compensation = 0.2
> enable_dynamic_overhang_speeds = 0
> external_perimeter_acceleration = 0
> external_perimeter_extrusion_width = 0
> external_perimeter_speed = 25
> external_perimeters_first = 0
> extra_perimeters = 0
> extra_perimeters_on_overhangs = 0
> extruder_clearance_height = 20
> extruder_clearance_radius = 20
> extrusion_width = 0.45
> fill_angle = 45
> fill_density = 15%
> fill_pattern = lightning
> first_layer_acceleration = 300
> first_layer_acceleration_over_raft = 0
> first_layer_extrusion_width = 0.65
> first_layer_height = 0.2
> first_layer_speed = 20
> first_layer_speed_over_raft = 30
> fuzzy_skin = none
> fuzzy_skin_point_dist = 0.8
> fuzzy_skin_thickness = 0.3
> gap_fill_enabled = 1
> gap_fill_speed = 40
> gcode_comments = 0
> gcode_label_objects = firmware
> gcode_resolution = 0.0125
> gcode_substitutions =
> infill_acceleration = 600
> infill_anchor = 400%
> infill_anchor_max = 50
> infill_every_layers = 1
> infill_extruder = 1
> infill_extrusion_width = 0
> infill_first = 0
> infill_overlap = 23%
> infill_speed = 45
> inherits = 0.20mm Quality @4max <https://github.com/4max> Pro 2.0
> interface_shells = 0
[... 1822 lines cut ...]
> filament_cost = 25
> filament_density = 1.156
> filament_deretract_speed = nil
> filament_diameter = 1.75
> filament_load_time = 0
> filament_loading_speed = 28
> filament_loading_speed_start = 3
> filament_max_volumetric_speed = 0
> filament_minimal_purge_on_wipe_tower = 15
> filament_multitool_ramming = 0
> filament_multitool_ramming_flow = 10
> filament_multitool_ramming_volume = 10
> filament_notes = "config v7 from hartrusion.com"
> filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
> filament_retract_before_travel = nil
> filament_retract_before_wipe = nil
> filament_retract_layer_change = nil
> filament_retract_length = nil
> filament_retract_length_toolchange = nil
> filament_retract_lift = nil
> filament_retract_lift_above = nil
> filament_retract_lift_below = nil
> filament_retract_restart_extra = nil
> filament_retract_restart_extra_toolchange = nil
> filament_retra

Fatihhamircu, 2023-12-25:
Unporcessed gcode of sliced models is in the attachment

Fatihhamircu, 2023-12-25:
[Uploading leftNerve_printReady.zip…]()

Fatihhamircu, 2023-12-26:
LOG file from Palette pro 3 
[Uploading log_2023_12_26T15_13_56_404Z.zip…]()

Fatihhamircu, 2023-12-28:
> Would you mind attaching the full unprocessed code?
> […](#)
> On 25 Dec 2023, at 18:34, Fatihhamircu ***@***.***> wrote: UPDATE: My bundle is: `# generated by PrusaSlicer 2.7.1+win64 on 2023-12-25 at [...]

[Uploading molarToothAndNerve.zip…]()

tomvandeneede, 2023-12-29:
OK will realle neerd the unprocessed file.  Best way of generating that is by temporarely adding the following line to your config and
;P2PP SAVEUNPROCESSED
afterwards you will find an unprocessed file in the temp folder.  This file is the one I can do some tests on.

Regards

Fatihhamircu, 2023-12-29:
[Uploading .10716.zip…]()

tomvandeneede, 2024-01-02:
I have not been able to download any of your files.  they both show as uploading.  and you please try resending the unprocessed file so I can take a look?

Fatihhamircu, 2024-01-03:
> I have not been able to download any of your files. they both show as uploading. and you please try resending the unprocessed file so I can take a look?

[.10716.zip](https://github.com/tomvandeneede/p2pp/files/13819055/default.10716.zip)

Fatihhamircu, 2024-01-08:
> I have not been able to download any of your files. they both show as uploading. and you please try resending the unprocessed file so I can take a look?

I have uploeded the file you want. Is it a bug?

Fatihhamircu, 2024-01-20:
To be honest, I have asked the same question to the Mosaic palette manufacturer. However, from their responds, I felt they did not developed the software for two object inside each other. Are your post processor software so?

Best regards

Fatihhamircu, 2024-03-24:
Have you looked it up the associated files?