tomvandeneede/p2pp #183: Creality K1 (rooted) and Palette 3 Pro - Accessory mode issue

ilker-aktuna, 2023-12-28:
Hi,

I am trying to use my Creality K1 printer with Palette 3 Pro.
I've setup P2PP as described in Wiki.
I could not find accessory mode settings there (Am I missing something ?) , but I found the 
G-Code for it : ";P2PP ACCESSORYMODE_MAFX"
When I add this to the start code, I get this error:

```
Processing time 1.89s
Generating GCODE file: (temp location, PS will move) C:\Users\ilker\AppData\Local\Temp\.PrusaSlicer.upload.d2f3-38e6-3abb-2938
Generating Palette 3 output files
Generating PALETTE MAFX file: Marlin_Iris_0.2mm_PLA__1h23m.mafx
We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp
Error: [Errno 13] Permission denied: 'Marlin_Iris_0.2mm_PLA__1h23m.mafx'
Traceback Info:
File "p2pp.py", line 115, in
File "C:\Users\tomvandeneede\Documents\p2pp\_build_update_\p2pp\p2pp\mcf.py", line 1134, in p2pp_process_file
File "C:\Users\tomvandeneede\AppData\Local\Programs\Python\Python39\lib\zipfile.py", line 1239, in __init__
```

---
When I don't put the accessory mode config, P2PP successfully creates a mcfx file and sends to the Palette 3 Pro. 
So uploading seems to work. But not for mafx. What am I doing wrong ?

Also, my K1 is rooted and I can directly send gcode from Prusa to K1 for single color prints. If I use P2PP, will it also send gcode to the printer ?

tomvandeneede, 2023-12-28:
can you please send me the knprocessed gcode file (output from prusaslicer before the p2pp script has runthis way I can tale a look ans see what the problem is is

ilker-aktuna, 2023-12-28:
> can you please send me the knprocessed gcode file (output from prusaslicer before the p2pp script has run

where can I find this file ?

ilker-aktuna, 2023-12-28:
[.PrusaSlicer.upload.zip](https://github.com/tomvandeneede/p2pp/files/13784959/default.PrusaSlicer.upload.zip)
is it this file ?

ilker-aktuna, 2023-12-29:
> can you please send me the knprocessed gcode file (output from prusaslicer before the p2pp script has runthis way I can tale a look ans see what the problem is isOp 28 dec 2023 om 09:32 heeft ilker Aktuna ***@***.***> het volgende geschreven:

Did you have a chance to look ?

tomvandeneede, 2023-12-29:
not yet have to work during the day so little time

tomvandeneede, 2023-12-29:
Lookslike the output pattern y ou are setting will generate a mcfx file.
Can you please change that to .gcode instead of .mcsf
With the currecnt settings as far as I can see this should generate an MAFX file for use on the Palette3 as well as a gcode file that will be used for printing on the Creality printer.
The files you are sent are not the unprocessed ones but the temp files for processing.   You can add the following line to the config to save the unprocessed file (will get an unprocessed suffix in the file name):
;P2PP SAVEUNPROCESSED
Don't forget to remove it afterwards as this will build up files in the temp folder that are never removed

ilker-aktuna, 2024-01-07:
Thanks. Selecting gcode works fine. 
It sends the mafx file to Palette 3 Pro
What about the gcode ? which gcode file should I manually upload to my printer ? 
and I did not understand what the option ";P2PP SAVEUNPROCESSED" will do for me ? Why do I need it ?

tomvandeneede, 2024-01-07:
yes manually upload or use prusaslicer to upload the gcode to the printer… that should word as well

ilker-aktuna, 2024-01-07:
is there a way to upload the gcode from Prusa slicer to the printer ? Normally without p2pp Prusa slicer can upload to my printer but with p2pp settings the result is different.

tomvandeneede, 2024-01-07:
Should be the same if you name the output file as .gcode file 
There is nothing special about the upload than before.   Just load the gcode and the print should start.
Unfortunately I have no printer af my disposal to test so you will have to do some testing with different file names etc, but in general the upload form P2PP should not impact the possibility to upload the processed file to your printer after P2PP processing