tomvandeneede/p2pp #185: P2PP Project Pulls Error

dondec123, 2023-12-31:
Hi Tom,
I created a P2PP project using a single color only.  The intent was to have the Palette3P automatically splice in the next reel of filament in, after my 1st "short reel" ran out.

This seemed to seriously confuse P2PP.  It complains that my single material BRA is not in the list... but it is.  

I tried it with other filament setups.  P2PP doesn't even realize it can use the DEFAULT splice tuning value either. 

I tried turning OFF and ON the Wipe tower option in Print Settings - Multiple Extruders... doesn't influence the outcome.

A copy of P2PP error message, my PS Project file (and .mcfx export ) file are here in this OneDrive folder:
https://1drv.ms/f/s!Arf1YeMUEh5Gm8MDgm6ETwFZERPWqA?e=PDA3pE 

   Don

tomvandeneede, 2024-01-01:
To my understanding the palette hardware does not support single color prints so if you want to print in a single color you always have to add a small part of q different color.  Have not heard that this has changed so I believe it still applies

dondec123, 2024-01-02:
Thanks.  The FB Palette group specified the procedure.  No P2PP or even USB connection needed.  You just slice normally and send to the printer.  On Palette select START -> Custom -> Multi Spool.  The Palette3 menus guide you the rest of the way.