tomvandeneede/p2pp #186: Accessory Mode - Number of pings: 0

moberli, 2024-01-01:
Hi

It seems P2PP do not send the number of pings to the palette 3/ 3Pro.

`Print Summary
----------------------------------------

Number of splices: 59
Number of pings: 0
Total print length 12904.21mm

Inputs/Materials used:
----------------------
Input 1 6560.43mm - PET [######] White 
Input 2 3749.28mm - PET [######] Peru 
Input 3 1024.05mm - PET [######] Black 
Input 4 1570.44mm - PET [######] Lime 

Splice Information:
-------------------
Splice Offset = 40.00mm
Autoloading Offset = 0.00mm
0001 Input: 1 Location: 0.00mm length 1147.33mm
0002 Input: 2 Location: 1147.33mm length 151.49mm
0003 Input: 1 Location: 1298.82mm length 319.80mm
0004 Input: 2 Location: 1618.62mm length 157.78mm
0005 Input: 1 Location: 1776.40mm length 320.20mm
0006 Input: 2 Location: 2096.60mm length 160.70mm
0007 Input: 1 Location: 2257.30mm length 313.37mm
0008 Input: 2 Location: 2570.66mm length 164.38mm
0009 Input: 1 Location: 2735.05mm length 310.53mm
0010 Input: 2 Location: 3045.57mm length 170.47mm
0011 Input: 1 Location: 3216.04mm length 293.25mm
0012 Input: 2 Location: 3509.29mm length 162.88mm
0013 Input: 3 Location: 3672.17mm length 167.06mm
0014 Input: 1 Location: 3839.23mm length 219.28mm
0015 Input: 2 Location: 4058.51mm length 195.91mm
0016 Input: 1 Location: 4254.43mm length 214.81mm
0017 Input: 3 Location: 4469.23mm length 171.16mm
0018 Input: 1 Location: 4640.40mm length 212.23mm
0019 Input: 2 Location: 4852.63mm length 203.41mm
0020 Input: 1 Location: 5056.04mm length 213.44mm
0021 Input: 3 Location: 5269.48mm length 173.20mm
0022 Input: 1 Location: 5442.68mm length 215.75mm
0023 Input: 2 Location: 5658.43mm length 213.94mm
0024 Input: 1 Location: 5872.36mm length 210.94mm
0025 Input: 3 Location: 6083.30mm length 171.82mm
0026 Input: 1 Location: 6255.12mm length 199.42mm
0027 Input: 2 Location: 6454.54mm length 225.82mm
0028 Input: 1 Location: 6680.36mm length 197.63mm
0029 Input: 3 Location: 6877.99mm length 170.55mm
0030 Input: 1 Location: 7048.54mm length 206.98mm
0031 Input: 2 Location: 7255.52mm length 202.73mm
0032 Input: 4 Location: 7458.25mm length 167.13mm
0033 Input: 1 Location: 7625.38mm length 211.22mm
0034 Input: 2 Location: 7836.61mm length 206.28mm
0035 Input: 3 Location: 8042.88mm length 170.26mm
0036 Input: 1 Location: 8213.14mm length 207.36mm
0037 Input: 2 Location: 8420.51mm length 211.00mm
0038 Input: 4 Location: 8631.50mm length 171.21mm
0039 Input: 1 Location: 8802.71mm length 196.54mm
0040 Input: 2 Location: 8999.26mm length 263.18mm
0041 Input: 1 Location: 9262.43mm length 183.77mm
0042 Input: 4 Location: 9446.20mm length 171.30mm
0043 Input: 1 Location: 9617.50mm length 176.62mm
0044 Input: 2 Location: 9794.12mm length 258.28mm
0045 Input: 1 Location: 10052.39mm length 174.63mm
0046 Input: 4 Location: 10227.02mm length 171.19mm
0047 Input: 1 Location: 10398.21mm length 173.43mm
0048 Input: 2 Location: 10571.63mm length 241.81mm
0049 Input: 1 Location: 10813.45mm length 171.89mm
0050 Input: 4 Location: 10985.34mm length 171.11mm
0051 Input: 1 Location: 11156.45mm length 170.53mm
0052 Input: 2 Location: 11326.97mm length 209.87mm
0053 Input: 1 Location: 11536.85mm length 169.92mm
0054 Input: 4 Location: 11706.77mm length 171.07mm
0055 Input: 1 Location: 11877.84mm length 169.58mm
0056 Input: 2 Location: 12047.42mm length 178.88mm
0057 Input: 4 Location: 12226.29mm length 170.94mm
0058 Input: 2 Location: 12397.24mm length 170.48mm
0059 Input: 4 Location: 12567.72mm length 376.49mm

Ping Information:
-----------------`

tomvandeneede, 2024-01-02:
can you please send me the full files you are docnloading tot he printer and to the P3

moberli, 2024-01-02:
[ping_issue_sidewipe.zip](https://github.com/tomvandeneede/p2pp/files/13811361/ping_issue_sidewipe.zip)

As requested, sorry what I missed to write is, this occurs if I use side wipe.

Custom G-code settings for p2pp

`
;P2PP PALETTE3_PRO

;P2PP ACCESSORYMODE_MAFX

;P2PP PRINTERPROFILE=50325050494e464f50325050494e464f

;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=80

;P2PP EXTRAENDFILAMENT=150

;P2PP LINEARPINGLENGTH=350

;P2PP SIDEWIPELOC=X0
;P2PP SIDEWIPEMINY=160
;set SIDEWIPEMAXY to the same value as SIDEWIPEMINY
;P2PP SIDEWIPEMAXY=160
;set WIPEFEEDRATE is required to override default value!!
;P2PP WIPEFEEDRATE=200

;P2PP MATERIAL_DEFAULT_0_0_0
`