tomvandeneede/p2pp #187: Times differ slicer & Mafx file

Blackcatswiskas, 2024-01-07:
I have an issue with accessory mode and timings. My sliced file should take over 10 days to print with over 3000 splices. The files run via p2pp from prusaslicer with no errors and provide the gcode and Mafx files however when I transfer the Mafx file via usb stick to the palette 3 it shows only 7 hours something (splice number seems around the same). I am unsure what the issue is. 
The link to download my files gcode and Mafx files 
[https://1drv.ms/u/s!AnVxWOV-WiWKhvxeqC024EhSsrgtWA?e=hiMJzf](url)

Blackcatswiskas, 2024-01-14:
Ok so I thought I’d start again from fresh putting all the setting in etc. I did so with a sheild small size in 4 colours sliced and put onto usb. Everything matched times solved etc on the gcode and MAFX file so I then tried scaling up to the full size item which fits on my 500x500 bed I DIDN’T change ANY settings slices using prusaslicer and p2pp loaddd onto usb and lo and behold the times differed but not the splices. I thought ok well let’s run this file see what happens all seemed to be printing fine until sometime later when I returned to the pallet to see that the print had indeed failed and the palette completely froze I couldn’t get anything to respond. See pictures this is the state the palette was stuck in the printer was in continuous motion making all the actions of continuing to print without being able to pull through any filament as it was locked by the palette.  In the end I had to physically pull the plug to reset it.   I’m not sure where the fault lies. But could someone please look into this Thanks I can provide the gcode files 

https://github.com/tomvandeneede/p2pp/assets/125461460/a953dc04-6645-47c6-b758-ed68a007721c

  

https://github.com/tomvandeneede/p2pp/assets/125461460/57ca8700-097c-4274-8b80-921491489713

[image]
[image]

Blackcatswiskas, 2024-01-14:
I am
Now running the print in one colour sliced by prusaslicer bypassing the palette altogether but obviously wish to use it for future prints

tomvandeneede, 2024-02-02:
Currently as far as I know P3 does not support single color prints.

Blackcatswiskas, 2024-02-03:
I think you have read me wrong the issue I have is with multi colour. I sliced and bypassed the p3 to try the file. Which although had the correct timings failed as it shifted on the x/y axis on several points which may or maynkot be related to the error as the first poisitiion shift is around where the file fails when multicolour showing incorrect timings. I am still at a loss as to whether it is the prusaslicer at fault or the p2pp and the shift is just another issue entirely. Any help would be appreciated (the later shifts a distance one way then two or three layers later it shifts back. Then does this again until it failed on a big shift near the top. )