tomvandeneede/p2pp #188: P2PP  generates error

cinealfa, 2024-01-21:
[image]
I am getting this error what am I doing wrong?

tomvandeneede, 2024-01-22:
can you please send me a copy of the file shown there ( /var/…..) and I will have a look

Screenshot.2024-01-21.at.15.28.14.png (view on web)

cinealfa, 2024-01-24:
Here you are!
[PrusaSlicer.upload.dfac-4a5d-8a7f-63c2.txt](https://github.com/tomvandeneede/p2pp/files/14032786/PrusaSlicer.upload.dfac-4a5d-8a7f-63c2.txt)

cinealfa, 2024-01-27:
Any thoughts?

tomvandeneede, 2024-01-27:
looks like the file is not in utf-8 formatwhat version of ps are you using?can you send me the filementioned in the screenshot (txt file)

cinealfa, 2024-01-27:
i did it is in the message right after you asked me to send it, Did I misunderstand you?

tomvandeneede, 2024-01-27:
yep got that but that was not a code file at all so this is definitely something P2PP cannot work with.
Which version of Prusa Slicer are you using?   Maybe there is a new file format that gets used now that P2PP is not (yet) aware of.

￼

Would be interesting to understand if this is the file BEFORE processing or AFTER
if after processing has been going on it would definitely be good to get a copy of the file BEFORE the processing by P2PP

tomvandeneede, 2024-01-27:
Looking at the actual error it appears to be happening in the reading of the file

￼

so the document you send is the input file from purse….

Can you please let me know what you have under printer settings/general and then the code flavour a thumbnail settings?  
I am expecting all data to be text encoded but in your file this seems no longer the case

￼

cinealfa, 2024-01-27:
Thank you for the assistance,  I m using 2.7.1 for OSX I will get to that computer and output the settings and check the flavor (which I believe is Klipper) and report back.

tomvandeneede, 2024-02-02:
Don't believe this has anything to do with the MACOs.   I also use MAC and have no issues.  The main problem seems to be the output file is not an uft-8 based text file but contains other data and P2PP is not able to process it

cinealfa, 2024-02-03:
What would be causing this? I I swear I followed the instructions in the PDF to the letter. Should I uninstall and start over? I haven't been tinkering under the hood with any of the 3-D printing stuff I use so I'm a little bit at a loss what I'm doing wrong.

sorgens, 2025-04-01:
I’ve just noticed that turning off ‘binary G-code’ resolves the situation.