tomvandeneede/p2pp #189: Missing documentation for `FIRMWARE_PURGE_LENGTH`

cyberhuman, 2024-01-22:
The colors in my first print with P2PP came out a bit skewed because there is a firmware purge and the parameter `FIRMWARE_PURGE_LENGTH` wasn't documented. It would be nice to at least mention it in the documentation.
The setting was deemed important enough by the Mosaic team to be added in their Canvas slicer:
[image]

Thank you for this project!

tomvandeneede, 2024-02-02:
For which printer is this used?   Is it just an amount of material added to the first splice before the printing starts?  Just would like to understand how to implement

cyberhuman, 2024-02-03:
I tested it and it works. I can also confirm the extrusion and splice lengths are calculated correctly because I had an unrelated issue and checked everything thoroughly.

tomvandeneede, 2024-02-03:
will add it next time I make an update

closed 2024-02-03 by tomvandeneede