tomvandeneede/p2pp #191: P2PP Generated .mcfx files lead to error "ENOENT"

5hiftyy, 2024-02-10:
I'm working on setting up my Ender 3 (Biqu H2 hotend, Klipper) with a Palette 3 Pro. I've tried PrusaSlicer 2.7.1 and 2.6.1 with P2PP 9.01.05 and 9.01.04, and followed the documents [Here](https://github.com/tomvandeneede/p2pp/blob/master/docs/P2PP%20P3%20Setup.pdf) probably three times through. However, I have two errors I believe I'm running into. 

The first, is the .mcfx file that P2PP generates, always produces the "ENOENT" error on the P3P. There's zero documentation on this error from Mosaic, either. I can only find one other reference to the error, and that's a [Facebook post](https://www.facebook.com/groups/130130450956865/search/?q=enoent) that no one replied to, so I'm kid of stuck. 

[image]

I then went and tried generating a file through Canvas, and the P3P fired right up. This leads me to believe there is an issue somewhere in the P2PP export post-process that's wrong, but I can't figure out what, as the slicing completes without issue or warning. 

The second, is I'm trying to figure out how to run my printer with Klipper with the P3P in accessory mode. It won't accept the .mcfx file as a .gcode substitute, (at least not the ones I'm currently generating) and P2PP only generates one file; the .mcfx, and no .gcode file for the printer, despite the ;P2PP ACCESSORYMODE_MAFX modifier in my start GCODE. 

Speaking of which, here is the entirety of my P2PP start Gcode: 

> ;P2PP PALETTE3_PRO
> ;P2PP P3_PROCESSPREHEAT
> ;P2PP P3_UPLOADFILE
> ;P2PP P3_HOSTNAME=192.168.0.77
> ;P2PP PRINTERPROFILE=39b434ce2de5b3e57229fad939839952
> ;P2PP LINEARPINGLENGTH=350
> ;P2PP SPLICEOFFSET=30
> ;P2PP MINSTARTSPLICE=130
> ;P2PP MINSPLICE=90
> ;P2PP ABSOLUTEEXTRUDER
> ;P2PP MATERIAL_DEFAULT_2_2_3
> ;P2PP MATERIAL_PLA_PLA_4_2_3
> ;P2PP MATERIAL_ABS_ABS_6_2_4
> ;P2PP MATERIAL_PLA1_PLA2_2_2_3
> ;P2PP MATERIAL_PLA2_PLA1_2_2_3
> ;P2PP MATERIAL_PET_PET_3_2_4
> ;P2PP EXTRAENDFILAMENT=150
> ;P2PP CHECKVERSION
> ;P2PP ACCESSORYMODE_MAFX
> ;P2PP P3_SHOWPRINTERPAGE
> ;P2PP P3_ROTATEINTERFACE
> ;P2PP KLIPPER_TOOLCHANGE
> :P2PP SAVEUNPROCESSED
> :P2PP CONFIGEND

The files are also only 55KB in size, where I'm used to much larger .gcode files, so I'm wondering if P2PP is outputting any information at all?

5hiftyy, 2024-02-10:
[Debug.zip](https://github.com/tomvandeneede/p2pp/files/14227939/Debug.zip)

Processed and Unprocessed files here.

5hiftyy, 2024-02-11:
I found what was causing the issue. When I was exporting the files from PrusaSlicer, I had the export option "Save As Type" set to "G-code Files (*.mcfx, *.gcode, ...)" I changed that to "G-Code Files (*.gcode)" and got P2PP to open, process, and upload the file to the P3P, as well as exporting a Gcode file to upload to the printer. 

However, two more issues have popped up: The first colour prints at 5mm/s until the print gets to the purge tower. Once the print gets to the purge tower, however, it runs into an error "move exceeds maximum extrusion" and Klipper cancels the print. 

I've double-checked my start/end GCODES in Klipper to make sure it doesn't have a relative/absolute conflict with P2PP, but I can't find anything obvious. Investigation continues tomorrow.

tomvandeneede, 2024-02-11:
I’ll try to take a look asap