tomvandeneede/p2pp #193: p2pp not working any more

pertoft, 2024-04-11:
Hi,

I have been using p2pp for a very long time - and it just works! Great job!

Lately, I have experience a problem with my Palette2s running in connected mode with klipper. 

I slice using prusa slicer and p2pp, uploading it to moonraker and starts print. 
The palette starts making filament and gets ready asks for me to insert the filament into the extruder.

When I pull filament from the palette it will NOT make new filament, and when the buffer is empty i cant pull out anymore.
Initially I thought it was an hardware problem and contacted palette support. After some troubleshooting, a hardware problem is ruled out.

We discovered that the problem only occurs when the gcode is sliced with p2pp and prusa slicer. If the same settings and printer profile is sliced in canvas3d, everything works.

There seem to be something in the gcode produced by p2pp which makes the palette2 not feeding filament correct.

Video of the problem: https://photos.app.goo.gl/KDTP1US9gGorCBbi7

Attached gcode:
[working-print.gcode.txt](https://github.com/tomvandeneede/p2pp/files/14944238/working-print.gcode.txt)
[faulty-print.gcode.txt](https://github.com/tomvandeneede/p2pp/files/14944240/faulty-print.gcode.txt)

Any suggestion what to change? Bug?

tomvandeneede, 2024-04-12:
Did you mane any changes tot the process (new versions of Prusa Slicer e.g.)

pertoft, 2024-04-12:
I have not used my setup for a long time. I have updated Prusa slicer to latest 2.7.4 and single color prints works flawlessly.
I upgraded to the latest release on p2pp before last print. The slicing looks find and everything seems good.

The only problem is that the palette2 crashes / are slow / cpu overload and do not produce filament when its pulled from the buffer.

tomvandeneede, 2024-04-12:
can you please send me the unprocessed file 
Will have a look

pertoft, 2024-04-13:
Do you mean the gcode after p2pp or the STL?

tomvandeneede, 2024-04-13:
Code file without the p2pp processing (temp disable the p2pp plugin but keep all the other settings the same)

pertoft, 2024-04-17:
[201901MosaicKeychain1_0.2mm_PLA_CR10S_51m.mcf.p2pp.gcode.txt](https://github.com/tomvandeneede/p2pp/files/15008407/201901MosaicKeychain1_0.2mm_PLA_CR10S_51m.mcf.p2pp.gcode.txt)
[201901MosaicKeychain1_0.2mm_PLA_CR10S_51m.mcf.plain.gcode.txt](https://github.com/tomvandeneede/p2pp/files/15008408/201901MosaicKeychain1_0.2mm_PLA_CR10S_51m.mcf.plain.gcode.txt)