tomvandeneede/p2pp #194: No Pings Detected

cdsimpkins, 2024-04-14:
Palette Device: Palette 3 Pro (Accessory Mode, latest firmware)
Printer: Ender 3 Pro (Marlin v1.3.1, 2020-04-25)
Slicer: PrusaSlicer (Latest -- Have updated at least once in the last month) 
OS: Mac OS Ventura 13.6

Processed, Unprocessed, and Log files [Here](https://gist.github.com/cdsimpkins/57e1edfff1228bc73b0826496758008b)

I've been working with my palette 3 for about a month, slowly trying to get it tuned and calibrated.  It's caused me to buy some replacement parts for both it and my printer here and there.  I now have the calibration print working just fine. 

My First Layer (and first transition) looks good to me.  
[image]

My mid-print transition looks perfect:
[image]
My final print looks perfect to me: 
[image]

My problem is that though there are several pings in the print: 
> Ping Information:
> -----------------
> Ping 0001 at 371.40mm
> Ping 0002 at 722.20mm
> Ping 0003 at 1098.08mm
> Ping 0004 at 1473.99mm
> Ping 0005 at 1819.50mm
> Ping 0006 at 2170.10mm
> Ping 0007 at 2530.48mm

The palette never detects any: 
[image]

I have definitely seen the printer do the pauses for Pings.  I've even noticed that there are pings that occur after the Palette is finished making filament (and I assume no longer looking for Pings)

I have tried all of these:  
- Calibrating Esteps
- [Calibrating Loading Offset](https://support.mosaicmfg.com/Wiki/Palette_3_Loading_Offset_and_Historical_Modifier)
- Switching to a geared extruder (and Calibrating Esteps)
- Cleaning the Palette (I've noticed filament debris stuck to the scroll wheel)
- Several different multi-color test models (larger and smaller) 

I'd really like to get Accessory mode working so my Palette can be agnostic of any future changes in printers.  I plan to eventually move the printer to Klipper firmware but I'm trying to get all this sorted out first.  I'm not sure it's a problem with p2pp.  I have not tried using Canvas, yet.  I've been avoiding it.  I suppose that could be my next troubleshooting step but I was hoping someone here would have some insight.  

I suppose I could also try Connected Mode and see if that works better, but I really want Accessory mode to work so that way the Palette could remain truly agnostic of whatever printer I have attached to it.

tomvandeneede, 2024-04-14:
Would you mind sending me the unprocessed GCODE file so I can have a look and do some tests myself.

Regards

cdsimpkins, 2024-04-14:
Sure,  it was in the gist but I guess git didn't make that very clear.  [Here's a direct link](https://gist.githubusercontent.com/cdsimpkins/57e1edfff1228bc73b0826496758008b/raw/8202001c5e13283dc92314b5f3f0509ca0df6434/unprocessed.gcode)

cdsimpkins, 2024-04-25:
Files: 
[Gcode Sliced by Canvas](https://gist.github.com/cdsimpkins/4fd590b5bb23496c59dccfc9c1ae6e5a#file-canvas201901mosaickeychain1-gcode)

I was able to get back to this today and try Canvas.  I imported my base printer config from Prusa Slicer and took out all my custom gcode to just get a base print going.  I still ran the palette in Accessory mode. 

Palette Device: Palette 3 Pro (Accessory Mode, latest firmware)
Printer: Ender 3 Pro (Marlin v1.3.1, 2020-04-25)
Slicer: Mosaic Canvas
OS: Mac OS Ventura 13.6

While I think the print quality is terrible (Most likely due to not really tuning the profile), the transitions still happened at the appropriate times and I didn't see any color bleed due to bad transitions: 
[image]

My Palette was able to detect pings.  I did notice that they were all at ~98%, which means it's using about 2% less filament than expected to get to all the pings, right? 
[image]

I also noticed that the printer still did pings after the palette was finished making filament, so I guess that is expected behavior? 

Is this information helpful?  Is there anything else I can provide?