tomvandeneede/p2pp #195: getting an error

wizard402, 2024-04-28:
this is the error im getting

P2PP Version: 9.01.05
Python Version: 3.9.1
Platform: win32

Reading File C:\Users\Lemieux\AppData\Local\Temp\.44084.gcode.pp
Analyzing Prusa Slicer Configuration

We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: invalid literal for int() with base 10: '113.75'

Traceback Info:
File "p2pp.py", line 115, in 
File "C:\Users\tomvandeneede\Documents\p2pp\_build_update_\p2pp\p2pp\mcf.py", line 1034, in p2pp_process_file 
File "C:\Users\tomvandeneede\Documents\p2pp\_build_update_\p2pp\p2pp\psconfig.py", line 440, in parse_config_parameters

tomvandeneede, 2024-04-29:
would you mind sending me the unprocessed code file so I can try to see what is happening

SimplifyAndAddCoffee, 2024-09-21:
I am getting the *exact* same error.  See attached.
[kinesis-maltron-dactyl_legends_0.1mm_PLA_MK4IS_19h14m.mcfx.zip](https://github.com/user-attachments/files/17083104/kinesis-maltron-dactyl_legends_0.1mm_PLA_MK4IS_19h14m.mcfx.zip)