tomvandeneede/p2pp #198: Orca slicer and P2PP

mikepersson, 2024-07-22:
i like orca slicer and want it to work with P2PP. Been in contact with orca and they say the problem lies with P2PP
[image]

tomvandeneede, 2024-07-22:
Looks like you have the priming of all extruders still turned on and you ar missing some of the config.
Did you follow the installation instructions?

Regards

mikepersson, 2024-07-23:
are there installation instructions to Orca?

tomvandeneede, 2024-07-23:
There are no specifics fir orca slicer but I expect the to be similar being a prusaslicer variant

mikepersson, 2024-07-23:
I contacted orca and they said it should be something P2PP could fix easily.

tomvandeneede, 2024-07-23:
Did they give you any more specifics?
I really stopped working on p2pp but willing to make an effort but I am not familiar with orca so any help would be welcome

mikepersson, 2024-07-23:
SoftFever
OP
 — 21/07/2024 10:17
As I said, that application doesn't support Orca, it's NOT an Orca problem.

I have it working in Prusa

Avatarsia, 2024-08-23:
> Did they give you any more specifics? I really stopped working on p2pp but willing to make an effort but I am not familiar with orca so any help would be welcome
> […](#)

what you will need to fix that ? 
i try myself to run a p3p with orca atm but running in the following errors 
[Print Settings][Multiple Extruders][Wipe Tower]Prime all printing extruders MUST be turned off
THIS FILE WILL NOT PRINT CORRECTLY
The PRINTERPROFILE identifier is missing, Default will be used 50325050494e464f 
SPLICE_OFFSET not defined

for splice offset i tried with and without _ same result. 

start gcode

;P2PP PALETTE3_PRO
;P2PP P3_PROCESSPREHEAT
;P2PP P3_UPLOADFILE
;P2PP P3_HOSTNAME=192.168.0.66
;P2PP P3_SHOWPRINTERPAGE
;P2PP P3_ROTATEINTERFACE
;P2PP PRINTERPROFILE=f22d46c50134cf47bdbc17eacaa47842
;P2PP LINEARPINGLENGTH=350
;P2PP SPLICE_OFFSET=40
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_0_0_0
;P2PP MATERIAL_ABS_ABS_0_0_0
;P2PP MATERIAL_PLA1_PLA2_0_0_0
;P2PP MATERIAL_PLA2_PLA1_0_0_0
;P2PP MATERIAL_PET_PET_0_0_0
;P2PP EXTRAENDFILAMENT=150

[image]
the 2.d Point Manual Multimaterialchange have no effekt at all
the multimaterial filament settings also seems to have no effekt, ramming is also to set 0 
[image]

CNCXPROUK, 2024-10-02:
It seems to work better of sorts with the new version of Orca However, it continues to give the following errors: 

SHORT FIRST SPLICE (min 100mm) Length:63.43 Input 0
The PRINTERPROFILE identifier is missing, Default will be used 50325050494e464f 
WARNING: No Algorithm defined for transitioning PLA to PLA. Using Default
Processing time 2.97s
Generating GCODE file: (temp location, PS will move) /var/folders/yr/b5d9nq5910n1cbb1p4_yssxr0000gn/T/orcaslicer_model/Wed_Oct_02/16_16_10#14600#93/Metadata/.14600.0.gcode.pp
SPLICE_OFFSET not defined

I compared the output from Prusa and Orca and the start of the Gcode is the same. I can't understand why its not working