tomvandeneede/p2pp #208: P2PP Layer Configuration is missing!!

Core3dTech, 2025-02-23:
Both myself and another are reporting this on the FB page with no real solution offered.

Setup P2PP in Prusa Slicer. The Configuration looks in order but still getting the same error.

[image]

[image]

[image]

tomvandeneede, 2025-02-24:
> Core3dTech
>  created an issue

Looks like you have an issue with the code generation.   Both lines in the generates code miss the ; so they are not picked up by the parser
Make sure you have the semicolon in front of each of these lines in the PS setup

bdavidhicks, 2025-02-24:
I think you are right about the missing semicolons for @Core3dTech but I am the other person he was referring to having this issue. I was able to run the python code on my machine after some trial and error with the python packages. I am getting the same error message but for a different reason which I believe I figured out. I will open a new issue now.

Core3dTech, 2025-02-24:
I'll have a second look but the screen dump shows the semicolons in my setup present.

bdavidhicks, 2025-02-24:
@Core3dTech see my issue #209 I just made for the root cause relating to my Z Offset Printer setting not being part of the layer calculations properly.

Core3dTech, 2025-02-24:
@bdavidhicks You are absolutely right, once I changed my offset to 0.3 instead of 0.24 (0.3 is layer height) the generation completes. Thank You!!

bdavidhicks, 2025-02-24:
That's good to hear, just be careful because now you might have bed adhesion problems but I'm sure you know that 😎